QuickLogic-Corp/quicklogic-fpga-toolchain #6: `counter_16bit`, `KeyError: 'src.x'` in `place` step, in create_place_constraints.py, after installing with "1) run an installer"

tcal-x, 2020-07-03:
On a new Ubuntu 18.04 LTS VM, I used the installer method, and installed in my home directory without using `sudo`.

In the `counter_16bit` directory, I ran
```
ql_symbiflow -compile -d ql-eos-s3 -P pd64 -v counter_16bit.v -t top -p counter_16bit.pcf 
```
according to instructions.  In the `place` step, I got the error:
```
Traceback (most recent call last):
  File "/home/tcal/opt/eos-s3/install/bin/python/create_place_constraints.py", line 183, in <module>
    main()
  File "/home/tcal/opt/eos-s3/install/bin/python/create_place_constraints.py", line 53, in main
    int(row["src.x"]),
KeyError: 'src.x'
make: *** [Makefile.symbiflow:21: build/top.place] Error 1
```

The contents of `row`:
```
row: OrderedDict([('name', 'B1'), ('x', '4'), ('y', '3'), ('z', '0'), ('type', 'BIDIR')])
```

tcal-x, 2020-07-03:
`top_io.place` contains the line:
```
#Block Name    x   y   z    pcf_line
#------------- --  --  -    ----
$iopadmap$top.count_4   4   3  0  # set_io count(4) B1
```

kkumar23, 2020-07-03:
@tcal-x : This is now fixed

tcal-x, 2020-07-03:
@kkumar23 -- Thanks, I am now able to generate a bitstream.

closed 2020-07-03 by tcal-x