QuickLogic-Corp/quicklogic-fpga-toolchain #9: Instructions for compiling from source are wrong

whitequark, 2020-07-11:
No such branch:

```
$ git clone https://github.com/SymbiFlow/vtr-verilog-to-routing -b blackbox_timing
Cloning into 'vtr-verilog-to-routing'...
fatal: Remote branch blackbox_timing not found in upstream origin
```

mithro, 2020-07-15:
Actually there are multiple issues with the "3) Compile from source code and run example" section at https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain#3-compile-from-source-code-and-run-example

Line 124 is using a C style comment in a bash shell terminal output - 
https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/blame/master/README.md#L124

Lines 136 and 137 have multiple errors. The line is part of a git clone command (it's missing the leading `git clone` bit), the comment doesn't match the line and as @whitequark pointed out the branch blackbox_timing doesn't exist on the SymbiFlow upstream repo;
https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/blame/master/README.md#L136-L137

kkumar23, 2020-07-15:
Fixed these issues

closed 2020-07-15 by kkumar23