QuickLogic-Corp/quicklogic-fpga-toolchain #13: Can't access IO_29 with PCF

timsaxe, 2020-08-05:
In pcf file I said 'set arnold_rst_o 29' and got an error message that said pad29 was not available

acomodi, 2020-08-06:
Hi @timsaxe,
Can you please provide an example to reproduce the error?

I think that the syntax may be wrong, the pcf file should have the following form:

```
set_io <io_name> <IOPAD name>
```

Also looking at the `pinmap_PU64.csv`, it appears that the pad 29 is indeed absent, I'll check whether this is expected.

timsaxe, 2020-08-06:
My error.  I had been looking at the quickfeather user guide, and it refers to IO_## so I assumed the # in the PCF were these numbers, but they are not, they are QFN pin # 

Raises the question: how might we avoid other users making the same mistake?

Thought #1: add a comment at the the top of the example pcf files reminding people what the #s refer to
Thought #2: add an EOSS3 'package' that uses the IO_# notation which would avoid the need for a decoder ring to map between the names used by the different worlds.

closed 2020-08-06 by timsaxe