QuickLogic-Corp/quicklogic-fpga-toolchain #14: PCF can;t access IO_11

timsaxe, 2020-08-05:
in PCF I had 'set arnold_rst_o 11' and got an error message something like trying to assign type PB-BIDIR to SDIOMUX

acomodi, 2020-08-06:
Hi @timsaxe,
Can you please provide an example to reproduce the error?

I think this is related to the fact that the block produced by the VPR packer that you are trying to LOC is not compatible with the PAD location you specified.

If possible, try to use a different pad, that is of type PB-BIDIR

timsaxe, 2020-08-06:
My error.  I interpreted the # in the pcf file as IO #, but they are QFN pin numbers.

Thoughts on how to avoid other people having the same confusion:
#1: is there are comment mechanism for pcf files because I’d like to put a note in mine saying these are QFN pin #s
#2: perhaps we should create an EOSS3 ‘package’ that uses IO_## notation, since that is the way the documentation looks and the way s/w describes it

timsaxe, 2020-08-06:
My error.  I had been looking at the quickfeather user guide, and it refers to IO_## so I assumed the # in the PCF were these numbers, but they are not, they are QFN pin # 

Raises the question: how might we avoid other users making the same mistake?

Thought #1: add a comment at the the top of the example pcf files reminding people what the #s refer to
Thought #2: add an EOSS3 'package' that uses the IO_# notation which would avoid the need for a decoder ring to map between the names used by the different worlds.

closed 2020-08-06 by timsaxe