QuickLogic-Corp/quicklogic-fpga-toolchain #15: Add an EOSS3 'package' that uses IO## nomenclature

timsaxe, 2020-08-07:
Second person just ran into this.  The software required you to know IO (or pad) numbers in order to configure the pads.  The Quickfeather UG describes available signals in terms of IO numbers.  All of which make the natural language for the pcf file to be IO numbers.  The board developer needs to know the IO number to pin/ball mapping, but that is a one-time activity.

timsaxe, 2020-08-07:
BTW, the naming of IO should be the same as in the data sheet.  So if the data sheet says IO_## then it should be IO_##, and if the datasheet says IO## then it should be IO##.  (Same should be true for the #defines in the .h file, but I'm pretty sure that is not the way it is today).

lnsharma, 2020-08-17:
IO naming is taken from the data sheet and different packages are created using the pin names as mentioned in the data sheet for the particular package. Please refer to the attached xls that captures the names for different packages. I sent this xls for review so that if there is any error then I can rectify it. Please let me know if you notice any issue in this xls. 
[QLAL4S3B_floorplan.xlsx](https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/files/5084005/QLAL4S3B_floorplan.xlsx)

timsaxe, 2020-08-17:
I guess I'm not being clear enough,.  You document is correct as far as I can tell, and follows the conventions of the FPGA world. where IO are mapped onto package pins.  This is necessary, obviously, for anyone dealing with the actual package.  And it works well for people who are comfortable with schematics.  But we are trying to make it easier for software engineers and lazy people like me to use the eFPGA.  So I was just building a piece of gateware to act as a tester so we can verify gateware when it is pout in the eFPGA. In order to do this I have to:

1. Look at the QF UG to find connectors -- these are well labeled and labeled with the IO_XX names from the datasheet
2. In order to create the PCF file I have to map them to the QFN pin numbers
* The lazy part of me says why do I have to map them?  Why can't I just use the IO_XX name?
* The other part of me goes and hunts down the schematic (takes time), chases the signals from the connectors (takes time and is error prone) and builds the necessary mapping table
* This took maybe 1/2 hour which is 1/2 hour longer than if I'd just been able to use IO_XX int eh PCF file

Now you might ask why not change the QF UG to use QFN pin numbers instead of IO_XX. That would avoid the mapping.  True, but I also have to update the pincfg.c file to indicate the properties (connect to FPGA, drive strength, etc) and that is done in terms of IO_XX because it has to work for all packages.  So if the QF UG uses QFN package names I still have to create a mapping table.

So my suggestion is that we create a new package called SW that uses the IO_XX naming.

Here we have a choice.  We can create one that has all IO and that means if the user specifies one that isn't bonded out that will not be flagged as an error and not bonded out.  Or we could create one per package, perhaps PU64SW, WR42SW and PD64SW.  I don;t have a strong opinion.  I have attached an updated XLS that has the single SW version as column AX.

BTW, perhaps a more elegant solution would be to allow aliases in the package descriptions, so for PU64 cell IO2 could be either 4 or IO_0.

BTW, this XLS also explains the other issue that I saw where IO_33 and greater don;t work, because the cell types switch from BIDIR to SDIOMUX.

[QLAL4S3B_floorplan-ts.xlsx](https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/files/5085867/QLAL4S3B_floorplan-ts.xlsx)

lnsharma, 2020-08-18:
The names you are referring to are the base names (mentioned in column AN). I think elegant solution would be to allow aliases in all 3 packages. As you mentioned, for example - in PU64 package, cell IO2 can refer to either 4 or IO_0.
Will plan to make this changes in code.

timsaxe, 2020-08-18:
Yes, that would be an elegant solution.

lnsharma, 2020-08-20:
Checked in the solution by adding alias name to all packages' pin names.

closed 2020-08-20 by lnsharma

timsaxe, 2020-08-20:
Perfect.  That is a great solution. Looking forward to using this.