QuickLogic-Corp/quicklogic-fpga-toolchain #18: Missing file init.mem

Thirsty2, 2020-08-24:
When building in a clean docker container that previously was able to build without error (a couple of weeks ago) there is now an error due to a missing init.mem file:

make[3]: Entering directory '/yosys-symbiflow-plugins/ql-iob-plugin/tests/sdiomux'
yosys -s script.ys

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+2406 (git sha1 7407b6c7, gcc 9.3.0-10ubuntu2 -fPIC -Os)

-- Executing script file `script.ys' --

1. Executing Verilog-2005 frontend: design.v
Parsing Verilog input from `design.v' to AST representation.
Warning: Yosys has only limited support for tri-state logic at the moment. (design.v:15)
Generating RTLIL representation for module `\top'.
Successfully finished Verilog frontend.

2. Executing SYNTH_QUICKLOGIC pass.

2.1. Executing Verilog-2005 frontend: /opt/symbiflow/eos-s3/bin/../share/yosys/quicklogic/cells_sim.v
Parsing Verilog input from `/opt/symbiflow/eos-s3/bin/../share/yosys/quicklogic/cells_sim.v' to AST representation.
Generating RTLIL representation for module `\inv'.
Generating RTLIL representation for module `\buff'.
Generating RTLIL representation for module `\logic_0'.
Generating RTLIL representation for module `\logic_1'.
Generating RTLIL representation for module `\gclkbuff'.
Successfully finished Verilog frontend.

2.2. Executing Verilog-2005 frontend: /opt/symbiflow/eos-s3/bin/../share/yosys/quicklogic/pp3_cells_sim.v
/opt/symbiflow/eos-s3/bin/../share/yosys/quicklogic/pp3_cells_sim.v:0: ERROR: Can not open file `init.mem` for \$readmemb.
make[3]: *** [Makefile:2: test] Error 1
make[3]: Leaving directory '/yosys-symbiflow-plugins/ql-iob-plugin/tests/sdiomux'
make[2]: *** [Makefile:9: sdiomux/ok] Error 2
make[2]: Leaving directory '/yosys-symbiflow-plugins/ql-iob-plugin/tests'
make[1]: Leaving directory '/yosys-symbiflow-plugins/ql-iob-plugin'
make[1]: *** [Makefile:20: test] Error 2
make: *** [Makefile:23: install_ql-iob] Error 2
The command '/bin/sh -c make' returned a non-zero code: 2

kkumar23, 2020-08-26:
@Thirsty2 : It is now fixed.

Thirsty2, 2020-08-26:
Thanks, I verified that everything builds again in the docker container.

closed 2020-08-26 by Thirsty2