QuickLogic-Corp/quicklogic-fpga-toolchain #19: make file_build_quicklogic_techmap_cells_sim.v does not work

Thirsty2, 2020-08-26:
When running the  step make file_build_quicklogic_techmap_cells_sim.v in a clean docker container after building the sources, I get:

root@6dee13d08c1f:/symbiflow-arch-defs/build/quicklogic# make file_build_quicklogic_techmap_cells_sim.v
make: *** No rule to make target 'file_build_quicklogic_techmap_cells_sim.v'.  Stop.

Thirsty2, 2020-08-26:
The following may be a result of the above, or a new issue.  When running 
root@6dee13d08c1f:/symbiflow-arch-defs/build/quicklogic/pp3/tests# make all_quick_tests, 
It proceeds until this point:
...
[  0%] Generating test_techmap_lut4_x100.vpp
Icarus Verilog Preprocessor version 11.0 (devel) (v10.0.0-853-g0a693031-dirty)

Copyright (c) 1999-2020 Stephen Williams (steve@icarus.com)

  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.
Using language generation: IEEE1364-2005,no-specify,xtypes,icarus-misc
PARSING INPUT
 ... done, 0.03 seconds.
ELABORATING DESIGN
/symbiflow-arch-defs/build/quicklogic/pp3/tests/techmaps/test_techmap_lut4_x100_dut_lut.v:7: error: Unknown module type: LUT4
/symbiflow-arch-defs/build/quicklogic/pp3/tests/techmaps/test_techmap_lut4_x100_dut_lut.v:15: error: Unknown module type: LUT4

Thirsty2, 2020-08-26:
On a different docker container (a different image that is closer to the yosys-hq Dockerfile) I still get the same error reported above when running make file_build_quicklogic_techmap_cells_sim.v. 

On this container, Running:
root@9f9c96866f4f:/symbiflow-arch-defs/build/quicklogic/pp3/tests# make all_quick_tests
Scanning dependencies of target file_quicklogic_pp3_tests_mult_mult.v
[  0%] Generating mult.v
...
proceeds until this point ...
Scanning dependencies of target test_techmap_lut2_x25
[  0%] Generating test_techmap_lut2_x25_dut_lut.v
[  0%] Generating test_techmap_lut2_x25_dut_mux.v
[  0%] Generating test_techmap_lut2_x25_miter.v
[  0%] Generating test_techmap_lut2_x25.vpp
Icarus Verilog Preprocessor version 11.0 (devel) (v10.0.0-853-g0a693031-dirty)

Copyright (c) 1999-2020 Stephen Williams (steve@icarus.com)

  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.
Using language generation: IEEE1364-2005,no-specify,xtypes,icarus-misc
PARSING INPUT
 ... done, 0 seconds.
ELABORATING DESIGN
/symbiflow-arch-defs/build/quicklogic/pp3/tests/techmaps/test_techmap_lut2_x25_dut_lut.v:7: error: Unknown module type: LUT2
/symbiflow-arch-defs/build/quicklogic/pp3/tests/techmaps/test_techmap_lut2_x25_dut_lut.v:13: error: Unknown module type: LUT2

mkurc-ant, 2020-08-31:
@Thirsty2 What version of SymbiFlow are you using? (which git SHA?). At some point the target for that file was renamed to `make file_build_quicklogic_pp3_techmap_cells_sim.v`. Besides that you shouldn't experience any issues with `make all_quick_tests` on the current master.

mkurc-ant, 2020-08-31:
@Thirsty2 Guess I was wrong. Targets for LUT techmaps fail when `all_quick_tests` is run. The fix should be easy though.

mkurc-ant, 2020-08-31:
This PR should fix that: https://github.com/QuickLogic-Corp/symbiflow-arch-defs/pull/148

Thirsty2, 2020-08-31:
@mkurc-ant Thanks for looking into this and for the PR.  I won't be able to test it until later tomorrow.  The attached dockerfile is what I used to run the builds.  It is based on the instructions found here:
https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain

Notice the 5th line from the bottom:
make file_build_quicklogic_techmap_cells_sim.v

This is the file that you mentioned has changed names, and the instructions in README.MD should be updated to reflect that, or perhaps changed entirely to run make all_quick_tests.  If no one has put in a fix to the instructions before I test your update tomorrow, I am happy to update the instructions and submit a PR for that tomorrow or Wed.
[Dockerfile.zip](https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/files/5151115/Dockerfile.zip)

Thirsty2, 2020-09-02:
I verified that 'make all_quick_tests' now works.  Thanks for fixing that.  The 5th line from the bottom of README.md is still wrong, but I am closing this issue and will post a PR to update the documentation to remove the line:
make file_build_quicklogic_techmap_cells_sim.v.

closed 2020-09-02 by Thirsty2