QuickLogic-Corp/quicklogic-fpga-toolchain #23: PCF file error

RandyO-QL, 2020-09-17:
under QuickLogic-Corp/s3-gateware i have a branch called USBserial_testqlsym12_RO. When attempting to compile with ql-symbiflow 1.2, i get this error:

PCF constraint "set_io spi_cs_o                11" from line 1 constraints net spi_cs_o of a block type PB-BIDIR to a location for block types:
PB-SDIOMUX
make: *** [Makefile.symbiflow:46: build/AL4S3B_FPGA_top.place] Error 1

Tim reported a similar error earlier with other I/O pins, so i think the same thing is happening to me here. It seems like it's finding a problem with the very first I/O pin that's listed, but please make sure the other I/O's in the quickfeather.pcf file will also work as well.

cibomahto, 2020-11-12:
I'm seeing this as well, it appears to affect all of the pins that are of type 'SDIOMUX'.

cibomahto, 2020-11-13:
Here's a minimal example that reproduces the issue for me, compiling with the 'v1.3.0: Quicklogic FPGA Toolchain Production Release':

https://gist.github.com/cibomahto/3294526a1047aa99cfacbf0e6894b0b8

It seems to happen regardless if the pin is set as input, output, or inout, and for all of these pins (from install/share/arch/ql-eos-s3_wlcsp/pinmap_PU64.csv):

22,3,31,0,SDIOMUX,IO_32
21,2,31,0,SDIOMUX,IO_33
20,1,31,0,SDIOMUX,IO_34
18,3,30,0,SDIOMUX,IO_35
17,2,30,0,SDIOMUX,IO_36
15,1,30,0,SDIOMUX,IO_37
16,3,29,0,SDIOMUX,IO_38
11,2,29,0,SDIOMUX,IO_39
13,1,29,0,SDIOMUX,IO_40
14,3,28,0,SDIOMUX,IO_41
10,2,28,0,SDIOMUX,IO_42
7,1,28,0,SDIOMUX,IO_43
8,3,27,0,SDIOMUX,IO_44
9,2,27,0,SDIOMUX,IO_45

RandyO-QL, 2020-11-13:
As a temporary workaround, until this is fixed, i believe we can use the actual device pin numbers instead of the FPGA port designations. Refer to the S3B device datasheet here:
https://www.quicklogic.com/wp-content/uploads/2020/06/QL-EOS-S3-Ultra-Low-Power-multicore-MCU-Datasheet.pdf
The pin table starts on page 95. Compiling in ql-symbiflow for a PU64 package will use the QFN package. So, looking at the 1st column, "signal name" gives the IO_xx FPGA port names. The QFN column gives the device pin numbers for the QFN package. If you use those device pin numbers in your .pcf file, i believe ql-symbiflow will work OK.
Example:
set_io my_signal 6  #IO_2

cibomahto, 2020-11-13:
trying by signal name:
`set_io BAD_PIN IO_45`

gives:
```
PCF constraint "set_io BAD_PIN  IO_45" from line 5 constraints net BAD_PIN of a block type PB-BIDIR to a location for block types:
PB-SDIOMUX
```

and trying by qfn pin number:
`set_io BAD_PIN 9`

gives:
```
PCF constraint "set_io BAD_PIN 9" from line 5 constraints net BAD_PIN of a block type PB-BIDIR to a location for block types:
PB-SDIOMUX
```

It looks like Yosys is incorrectly inferring a BIDIR block for these outpad/inpad/iopad entries, when it should infer an SDIOMUX.

mkurc-ant, 2020-11-17:
@cibomahto Have you tried to re-build the whole design? (i.e. remove the `build` directory and start over). I've noticed that when you change a pin in PCF after the design is already synthesized Yosys isn't re-run.

cibomahto, 2020-11-17:
@mkurc-ant Thanks, clearing the build directory does seem to get the workaround going (the alias still doesn't work on these pins though). It's unfortunate that the error message references the new pad name, even though it's not recompiling.

lnsharma, 2020-11-30:
I fixed the issue where alias pin names were not getting accepted for SDIOMUX. Please take the latest symbiflow to verify if it works for you.

closed 2020-11-30 by lnsharma