QuickLogic-Corp/quicklogic-fpga-toolchain #24: conda activate breaks tinyfpagprogrammer

timsaxe, 2020-09-20:
When I conda activate for symbiflow, tinyfpagprogrammer fails because the serial module is not available.  Please include serial in the symbiflow conda env and ensure that tinyfpgaprogrammer works.  It is cumbersome to have to conda active/deactivate between make a programming.

kkumar23, 2020-09-21:
@timsaxe : Sure Tim, please provide details on adding serial module.

timsaxe, 2020-09-21:
The way I did this was to ‘conda active’  then ‘pip3 install serial’

I believe our installer installs a bunch of modules, so I think the solution is to add serial to that list.  I have no idea how to do that.

kkumar23, 2020-09-22:
Thanks @timsaxe , i will take care of this.

kkumar23, 2020-10-16:
Fixed in Symbiflow v1.3.0

closed 2020-10-18 by kkumar23