QuickLogic-Corp/quicklogic-fpga-toolchain #26: Add instructions in the readme describing how to install/maintain/run 2 different versions of ql-symbiflow

RandyO-QL, 2020-09-25:
This may involve proper PATH manipulation and cleanup, conda activate/deactivate, and possibly opening a "fresh" console window to install or run a 2nd version. It would be nice if the steps to install/run a 2nd version were mentioned in the README.

Thirsty2, 2020-09-25:
One way to do it is to use a container.  It is possible to have many different versions of symbiflow in different containers.  I would be happy to add instructions on how to do this to the README.md, but I need to get better at editing .md files.  The addToREADME.md looked ok on my machine, and even looked OK on github at one time but it looks pretty bad when rendered in the browser via github now.  My intent was to add this to the RAEADME.md file, but I have never written an .md file until this one, and I didn't want to screw up the main README.md file.  If you can get past the poor formatting/rendering, it does describe how to use Docker to run symbiflow in a container.  I am happy to answer any questions.  I am also open to suggestions on how to best edit .md files so they look good on Github.

kkumar23, 2020-11-19:
As of now container support enables to run 2 different versions. So closing this issue.

closed 2020-11-19 by kkumar23