QuickLogic-Corp/quicklogic-fpga-toolchain #28: Having issue while install Sysmbiflow_v1.2.0

philquik, 2020-09-30:
Having issue while install Sysmbiflow_v1.2.0 now. It was working probably a week ago.
Error as below.
...
Collecting package metadata (current_repodata.json): failed

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://conda.anaconda.org/quicklogic-corp/label/ql/linux-64/current_repodata.json>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.
'https://conda.anaconda.org/quicklogic-corp/label/ql/linux-64'

Collecting package metadata (current_repodata.json): failed

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://conda.anaconda.org/quicklogic-corp/label/ql/linux-64/current_repodata.json>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.
'https://conda.anaconda.org/quicklogic-corp/label/ql/linux-64'

Collecting package metadata (current_repodata.json): failed

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://conda.anaconda.org/quicklogic-corp/linux-64/current_repodata.json>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.
'https://conda.anaconda.org/quicklogic-corp/linux-64'

Collecting package metadata (current_repodata.json): failed

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://conda.anaconda.org/tfors/linux-64/current_repodata.json>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.
'https://conda.anaconda.org/tfors/linux-64'

Collecting package metadata (current_repodata.json): failed

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://conda.anaconda.org/quicklogic-corp/label/ql/linux-64/current_repodata.json>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.
'https://conda.anaconda.org/quicklogic-corp/label/ql/linux-64'

WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f3532388c10>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/python-constraint/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f3531fb4190>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/python-constraint/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f3531fb42d0>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/python-constraint/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f3531fb4450>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/python-constraint/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f3531fb4590>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/python-constraint/
ERROR: Could not find a version that satisfies the requirement python-constraint (from versions: none)
ERROR: No matching distribution found for python-constraint
Collecting git+https://github.com/QuickLogic-Corp/quicklogic-fasm
  Cloning https://github.com/QuickLogic-Corp/quicklogic-fasm to /tmp/pip-req-build-6rahbnlm
Collecting fasm@ git+https://github.com/symbiflow/fasm#egg=fasm
  Cloning https://github.com/symbiflow/fasm to /tmp/pip-install-ey_7eefd/fasm
Collecting fasm-utils@ git+https://github.com/QuickLogic-Corp/quicklogic-fasm-utils#egg=fasm-utils
  Cloning https://github.com/QuickLogic-Corp/quicklogic-fasm-utils to /tmp/pip-install-ey_7eefd/fasm-utils
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f4ad3dbdf90>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/textx/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f4ad3d8db50>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/textx/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f4ad3d8db10>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/textx/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f4ad3d8de10>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/textx/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f4ad3da4910>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/textx/
ERROR: Could not find a version that satisfies the requirement textx (from fasm@ git+https://github.com/symbiflow/fasm#egg=fasm->quicklogic-fasm==0.0.1) (from versions: none)
ERROR: No matching distribution found for textx (from fasm@ git+https://github.com/symbiflow/fasm#egg=fasm->quicklogic-fasm==0.0.1)

kkumar23, 2020-09-30:
@philquik : Can you give another try, we have seen this when there is issue while connecting to anaconda. There is no change in the symbiflow installer though

philquik, 2020-10-05:
Still having problem.
If I used browser to click the 
https://conda.anaconda.org/quicklogic-corp/label/ql/linux-64/current_repodata.json
it is not found. Is it correct? Maybe the problem is the file on anaconda, not script?

kkumar23, 2020-10-05:
@philquik : Well the file on anaconda are fine. As we have installed and verified again. I see couple of solutions to this issue on few threads:
https://docs.conda.io/projects/conda/en/latest/user-guide/troubleshooting.html#condahttperror-http-000-connection-failed
https://github.com/conda/conda/issues/9746.

Please try this:
Temporarily set your ssl_verify variable to false, upgrade the requests package, and then set ssl_verify back to true using the following commands:
conda config --set ssl_verify false
_conda update requests_
conda config --set ssl_verify true

If that still does not work try conda clean
conda clean --all
conda config --set ssl_verify no

philquik, 2020-10-06:
Still no luck to make it work with above comments.
Is there a way I can download the necessary files and install locally?

Below is whole log. In the beginning, the Collecting package metadata (current_repodata.json) are all good.

==================================

phil@PHIL-ASUS-UX433:~$ bash Symbiflow_v1.2.0.gz.run
Verifying archive integrity...  100%   All good.
Uncompressing Symbiflow Conda Installer  100%
Installation starting for conda based symbiflow
QuickLogic Corporation
$INSTALL_DIR is set to /home/phil/Symbiflow_v1.2.0 !
--2020-10-06 14:51:31--  https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh
Resolving repo.continuum.io (repo.continuum.io)... 2606:4700::6812:c84f, 2606:4700::6812:c94f, 104.18.200.79, ...
Connecting to repo.continuum.io (repo.continuum.io)|2606:4700::6812:c84f|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh [following]
--2020-10-06 14:51:31--  https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh
Resolving repo.anaconda.com (repo.anaconda.com)... 2606:4700::6810:8303, 2606:4700::6810:8203, 104.16.131.3, ...
Connecting to repo.anaconda.com (repo.anaconda.com)|2606:4700::6810:8303|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 93052469 (89M) [application/x-sh]
Saving to: ‘conda_installer.sh’

conda_installer.sh            100%[=================================================>]  88.74M  2.61MB/s    in 37s

2020-10-06 14:52:08 (2.43 MB/s) - ‘conda_installer.sh’ saved [93052469/93052469]

PREFIX=/home/phil/Symbiflow_v1.2.0/conda
Unpacking payload ...
Extracting : ca-certificates-2020.1.1-0.conda:   3%|█                                    | 1/35 [00:00<00:09,  3.65it/sExtracting : certifi-2020.4.5.1-py38_0.conda:   3%|█                                     | 1/35 [00:00<00:09,  3.65it/sExtracting : certifi-2020.4.5.1-py38_0.conda:   6%|██▏                                   | 2/35 [00:00<00:07,  4.33itExtracting : cffi-1.14.0-py38he30daa8_1.conda:   6%|██                                   | 2/35 [00:00<00:07,  4.33it/Extracting : chardet-3.0.4-py38_1003.conda:   9%|███▍                                    | 3/35 [00:00<00:07,  4.33iExtracting : chardet-3.0.4-py38_1003.conda:  11%|████▌                                   | 4/35 [00:00<00:05,  5.65Extracting : conda-package-handling-1.6.1-py38h7b6447c_0.conda:  11%|██▎                 | 4/35 [00:00<00:05,  5.65itExtracting : cryptography-2.9.2-py38h1ba5d50_0.conda:  14%|████▎                         | 5/35 [00:01<00:05,  5.65Extracting : cryptography-2.9.2-py38h1ba5d50_0.conda:  17%|█████▏                        | 6/35 [00:01<00:06,  4.3Extracting : idna-2.9-py_1.conda:  17%|████████▌                                         | 6/35 [00:01<00:06,  Extracting : ld_impl_linux-64-2.33.1-h53a641e_7.conda:  20%|█████▊                       | 7/35 [00:01<00:06,  4.3Extracting : libedit-3.1.20181209-hc058e9b_0.conda:  23%|███████▎                        | 8/35 [00:01<00:06,  4Extracting : libffi-3.3-he6710b0_1.conda:  26%|██████████▊                               | 9/35 [00:01<00:05,Extracting : libgcc-ng-9.1.0-hdf63c60_0.conda:  29%|██████████▎                         | 10/35 [00:01<00:05,Extracting : libstdcxx-ng-9.1.0-hdf63c60_0.conda:  31%|██████████▎                      | 11/35 [00:01<00:05,Extracting : ncurses-6.2-he6710b0_1.conda:  34%|█████████████▋                          | 12/35 [00:04<00:Extracting : ncurses-6.2-he6710b0_1.conda:  37%|██████████████▊                         | 13/35 [00:04<00Extracting : openssl-1.1.1g-h7b6447c_0.conda:  37%|█████████████▋                       | 13/35 [00:04<00:Extracting : pip-20.0.2-py38_3.conda:  40%|██████████████████                           | 14/35 [00:04Extracting : pycosat-0.6.3-py38h7b6447c_1.conda:  43%|██████████████▌                   | 15/35 [00:04<00Extracting : pycparser-2.20-py_0.conda:  46%|███████████████████▋                       | 16/35 [00:Extracting : pyopenssl-19.1.0-py38_0.conda:  49%|██████████████████▉                    | 17/35 [00:0Extracting : pysocks-1.7.1-py38_0.conda:  51%|█████████████████████▌                    | 18/35 [0Extracting : python-3.8.3-hcff3b4d_0.conda:  54%|█████████████████████▏                 | 19/35 [0Extracting : python-3.8.3-hcff3b4d_0.conda:  57%|██████████████████████▎                | 20/35 [Extracting : readline-8.0-h7b6447c_0.conda:  57%|██████████████████████▎                | 20/35 [Extracting : requests-2.23.0-py38_0.conda:  60%|████████████████████████                | 21/35 Extracting : ruamel_yaml-0.15.87-py38h7b6447c_0.conda:  63%|█████████████████▌          | 22/35 [00:05Extracting : setuptools-46.4.0-py38_0.conda:  66%|████████████████████████▉             | 23/35Extracting : six-1.14.0-py38_0.conda:  69%|██████████████████████████████▊              |Extracting : sqlite-3.31.1-h62c20be_1.conda:  71%|███████████████████████████▏          | 25Extracting : tk-8.6.8-hbc83047_0.conda:  74%|███████████████████████████████▉           Extracting : tqdm-4.46.0-py_0.conda:  77%|███████████████████████████████████▍      Extracting : urllib3-1.25.8-py38_0.conda:  80%|████████████████████████████████▊       Extracting : wheel-0.34.2-py38_0.conda:  83%|███████████████████████████████████▋   Extracting : xz-5.2.5-h7b6447c_0.conda:  86%|████████████████████████████████████▊ Extracting : yaml-0.1.7-had09818_2.conda:  89%|████████████████████████████████████ Extractin [... cut ...]
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f86c3632110>: Failed to establish a new connection: [Errno -2] Name or service not known')': /simple/textx/
ERROR: Could not find a version that satisfies the requirement textx (from fasm@ git+https://github.com/symbiflow/fasm#egg=fasm->quicklogic-fasm==0.0.1) (from versions: none)
ERROR: No matching distribution found for textx (from fasm@ git+https://github.com/symbiflow/fasm#egg=fasm->quicklogic-fasm==0.0.1)

kkumar23, 2020-10-09:
@philquik : The issue can be solved by using the docker. It is under progress, once we have the support this issue wont occur.

kkumar23, 2020-10-22:
@kgugala : We see this issue comes up when the user has python 3.8, and we have created the yosys package with 3.7.
User runs the below command:
conda install -c quicklogic-corp/label/ql yosys="0.5_7972_g4045d484 20200817_053125" python=3.7

But if user skips the python version and installs with this command :
conda install -c quicklogic-corp/label/ql yosys="0.5_7972_g4045d484 20200817_053125"

Then another issue of solving environment pops up:

Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done

kgugala, 2020-10-22:
@kkumar23 on my local machine I'm using python 3.8.5 and I don't have any issues with installation. I checked this on toolchain release 1.2 and 1.3.

The original issue seems to be related to network connection: `CondaHTTPError: HTTP 000 CONNECTION FAILED`. Maybe conda server had a hiccup?

kkumar23, 2020-10-22:
@kgugala : Even i do not have issue on my local machines. But the user tried multiple times on his machine and the issue was re-occurring, so not related to conda server. 
So i did provide the step by step installation instead of toolchain release file. All steps were fine except when installing the yosys package when this error pops up.
@philquik : Add if i missed anything.

kgugala, 2020-10-22:
@philquik what Linux distro are you using?

philquik, 2020-10-22:
I am running WSL with ubuntu 18.04.

kgugala, 2020-10-22:
I suppose this is somehow related to WSL. I'll set up a test machine and take a look at this

kkumar23, 2020-10-22:
@kgugala : I tried a fresh install of Ubuntu WSL. and tried to compile the symbiflow-arch-defs: make env, is fine but make all_conda gives the below error:

PREFIX=/home/kkumar/symbiflow-arch-defs/env/conda
Unpacking payload ...
Collecting package metadata (current_repodata.json): done
Solving environment: done

# All requested packages already installed.

installation finished.
make[3]: env/conda/bin/conda: Command not found
CMakeFiles/file_env_conda_bin_conda.dir/build.make:79: recipe for target 'env/conda/bin/conda' failed
make[3]: *** [env/conda/bin/conda] Error 127
CMakeFiles/Makefile2:19522: recipe for target 'CMakeFiles/file_env_conda_bin_conda.dir/all' failed
make[2]: *** [CMakeFiles/file_env_conda_bin_conda.dir/all] Error 2
CMakeFiles/Makefile2:19319: recipe for target 'CMakeFiles/all_conda.dir/rule' failed
make[1]: *** [CMakeFiles/all_conda.dir/rule] Error 2
Makefile:481: recipe for target 'all_conda' failed
make: *** [all_conda] Error 2

kkumar23, 2020-10-23:
@kgugala : On both Ubuntu 18 WSL 18 and 20 WSL, The above error i fixed by creating the .conda folder. 
But i get stuck with the below message while compiling symbiflow-arch-defs or symbilfow installer:
I suspect the below yosys package does not go very well with conda environment. 
conda install -c quicklogic-corp/label/ql yosys="0.5_7972_g4045d484 20200817_053125" python=3.7

--------------------------------------------------------------
 Collecting package metadata (current_repodata.json): failed

CondaHTTPError: HTTP 000 CONNECTION FAILED for url <https://conda.anaconda.org/quicklogic-corp/label/ql/linux-64/current_repodata.json>
Elapsed: -

An HTTP error occurred when trying to retrieve this URL.
HTTP errors are often intermittent, and a simple retry will get you on your way.
'https://conda.anaconda.org/quicklogic-corp/label/ql/linux-64'

Collecting package metadata (current_repodata.json): failed

kkumar23, 2020-10-25:
> I suppose this is somehow related to WSL. I'll set up a test machine and take a look at this

@kgugala Did you get the similar issue on the windows WSL ?

kgugala, 2020-10-27:
@kkumar23 @philquik I've tested the 1.3 toolchain in WSL2 and it works without any issues. WSL version was ` (WSL2, Win10 ver. 2004, comp no. 19041.572) with Ubuntu 20.04LTS`

@philquik can you update WSL to the above version and try again with the latest toolchain?

philquik, 2020-10-27:
Great.  I will try that.

philquik, 2020-10-27:
@kgugala, Confirmed WSL2 working with ubuntu 18.04LTS. Thanks a lot.

kkumar23, 2020-10-27:
@kgugala @philquik : With fresh install and wsl2 update it fixed for me as well. Need to update the readme to have the wsl2 requirement.

closed 2020-10-27 by kkumar23