QuickLogic-Corp/quicklogic-fpga-toolchain #29: memory initialization files fail to be opened while synthesizing RAM blocks in "ram_test"

coolbreeze413, 2020-09-30:
Installed the toolchain using Symbiflow_v1.2.0 installer.
While running the compilation for "ram_test" directory in the installation, the read of memory initialization .hex files fails, because the tool cannot open the files. The tool tries to open the files in the build directory instead of the rtl sources directory, and hence it fails to open them.
How to reproduce:
1. Install Symbiflow_v1.2.0 using the release, and initialize according to steps.
2. Navigate to `$INSTALL_DIR/install/tests/ram_test`
3. Compile the design: `ql_symbiflow -compile -src . -d ql-eos-s3 -P PD64 -v *.v -t top -p chandalar.pcf -dump header`

In the generated `top_bit.h` the `axFPGAMemInit[]` array is empty.
In the synth logs, `build/top_synth.log` we see the following lines:
```
Processing u_AL4S3B_FPGA_IP.u_AL4S3B_FPGA_RAMs.RAM0_INST.RAM_INST : init_512x16.hex
Can not open file `init_512x16.hex`.
cell type u_AL4S3B_FPGA_IP.u_AL4S3B_FPGA_RAMs.RAM1_INST.RAM_INST
found ram block
Processing u_AL4S3B_FPGA_IP.u_AL4S3B_FPGA_RAMs.RAM1_INST.RAM_INST : init_1024x16.hex
Can not open file `init_1024x16.hex`.
cell type u_AL4S3B_FPGA_IP.u_AL4S3B_FPGA_RAMs.RAM2_INST.RAM_INST
found ram block
Processing u_AL4S3B_FPGA_IP.u_AL4S3B_FPGA_RAMs.RAM2_INST.RAM_INST : init_1024x8.hex
Can not open file `init_1024x8.hex`.
cell type u_AL4S3B_FPGA_IP.u_AL4S3B_FPGA_RAMs.RAM3_INST.RAM_INST
found ram block
Processing u_AL4S3B_FPGA_IP.u_AL4S3B_FPGA_RAMs.RAM3_INST.RAM_INST : init_512x32.hex
Can not open file `init_512x32.hex`.
cell type u_AL4S3B_FPGA_IP.u_AL4S3B_FPGA_RAMs.RAM4_INST.RAM_INST
found ram block
Processing u_AL4S3B_FPGA_IP.u_AL4S3B_FPGA_RAMs.RAM4_INST.RAM_INST : init_2048x8.hex
Can not open file `init_2048x8.hex`.
```

workaround: 
- After initial build, place the hex files from the `ram_test` directory into `build` directory
- Delete the `top.eblif`
- Re-run the compilation
we can see that now the hex files are picked up from the build directory, and the top_bit.h also contains axFPGAMemInit[] with valid values.

kkumar23, 2020-11-23:
@coolbreeze413 : this is now fixed , you can use the latest symbiflow script 1.3.0

coolbreeze413, 2020-11-23:
Thanks!
Can I install a fresh version using the same installer as in the releases?
https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/releases/download/v1.3.0/Symbiflow_v1.3.0.gz.run

kkumar23, 2020-11-23:
@coolbreeze413 Right. The hex files will now be copied to build folder before synthesis run.

coolbreeze413, 2020-11-23:
With latest 1.3.0 installer, the hex files get copied, and the ram.mem/memory initialization script generation happens correctly.

closed 2020-11-23 by coolbreeze413