QuickLogic-Corp/quicklogic-fpga-toolchain #37: Version 1.3.0 installer is broken

Thirsty2, 2020-10-22:
When building a docker container from the version 1.3.0 installer, many of the tools are missing, including synth, place, route and a few others that might be nice to have.

kkumar23, 2020-10-23:
@Thirsty2 All these are activated when below steps are added:
export INSTALL_DIR="specify the installpath"
#adding symbiflow toolchain binaries to PATH
export PATH="$INSTALL_DIR/install/bin:$INSTALL_DIR/install/bin/python:$PATH"
source "$INSTALL_DIR/conda/etc/profile.d/conda.sh"
conda activate

Thirsty2, 2020-10-23:
kkumar23,

The dockerfile performs the above steps, and I verified that the
environment vars INSTALL_DIR and PATH are set up correctly as above.  When
I run a bash shell interactively in the container I can source conda.sh,
but I cannot run synth, pack, place, etc.  I used 'find / -name synth' to
look for synth, and it doesn't show up.

The same Dockerfile runs fine with the 1.2.0 installer, and the tools exist
there.

Detailed steps:

Using the latest Dockerfile.use-installer (merged into the repo in the last
24 hours)

commit 514e2cd3ad6b2199f61c5252a5e69d3d389e9b76 (HEAD -> master,
origin/master, origin/HEAD)
Merge: 9f8920f bbd1721
Author: kkumar23 <61860672+kkumar23@users.noreply.github.com>
Date:   Fri Oct 23 16:09:51 2020 +0530

    Merge pull request #35 from Thirsty2/update-dockerfile-use-installer

    Update dockerfile use installer

build with:

docker build -f Dockerfile.use-installer . -t symbiflow-ql:1.3.0

Then run with:
docker run -it symbiflow-ql:1.3.0 bash

Inside the container, try:
root@5243f0b017b8:/# find / -name synth
root@5243f0b017b8:/# find / -name pack
root@5243f0b017b8:/# find / -name place

Notice that nothing is found.  Try the same thing with the 1.2.0 container:
Build the 1.2.0 container using the same dockerfile with:
docker build --build-arg TAG=v1.2.0 -f Dockerfile.use-installer . -t
symbiflow-ql:1.2.0

docker run -it symbiflow-ql:1.2.0 bash

Inside this container, repeat the searches for the tools:
root@2067795d50b5:/# find / -name synth
/opt/symbiflow/eos-s3/install/bin/synth
root@2067795d50b5:/# find / -name pack
/opt/symbiflow/eos-s3/install/bin/pack
root@2067795d50b5:/# find / -name place
/opt/symbiflow/eos-s3/install/bin/place
root@2067795d50b5:/#

Notice that they are found, whereas they were not found in the 1.3.0
container.

Doug

kkumar23, 2020-10-23:
Hi,
  We have changed the commands in 1.3.0 to symbiflow_<binary> . So synth will be symbiflow_synth .

Regards,
Kishor

Thirsty2, 2020-10-23:
Ok, I re-ran the 1.3.0 container and it does contain symbiflow_synth.  I
must have missed the memo...

Thanks,

Doug

Thirsty2, 2020-10-23:
The names of the synth, pack, place, route, etc binaries changed to symbiflow_synth, symbiflow_pack, symbiflow_place, etc.  I missed the memo.  The installer isn't broken.

closed 2020-10-23 by Thirsty2