QuickLogic-Corp/quicklogic-fpga-toolchain #38: failed to dial gRPC: cannot connect to the Docker daemon

kkumar23, 2020-10-23:
@Thirsty2 : I am not able to run the docker. 
Using ubuntu 18.04 WSL
Installer docker, restarted system.

Below command always gives error:
docker build . -f Dockerfile.use-installer -t symbiflow-ql

ERRO[0000] failed to dial gRPC: cannot connect to the Docker daemon. Is 'docker daemon' running on this host?: dial unix /var/run/docker.sock: connect: no such file or directory
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?

Thirsty2, 2020-10-23:
kkumar23,

Did you get this working again?  Is this a new machine, or is it the one
you were using before and it quit working?  I was going to suggest that you
uninstall and reinstall docker, but it looks like you may have tried that
already.  Are you using Windows 10 with WSL or WSL2?  I have a machine that
runs Docker under Windows 10 WSL2 with Microsoft's Linux kernel running
Ubuntu 20.04, and it works well for me, but I have been running native
Ubuntu 20.04 more often than not lately because the hardware USB drivers
seem to work better so I haven't tried it in a couple of weeks under WSL2.
I will do that to make sure everything still works in WSL2, then run
Windows updates to see if anything breaks.

You might check whether your firewall is blocking something as well.  I
wouldn't expect that to be a problem, but I don't know for certain.

Doug

kkumar23, 2020-10-25:
@Thirsty2 : Well I am just trying to validate the docker on windows wsl. But looks like the docker service itself is not getting enabled. Will check if there is any firewall issues!

coolbreeze413, 2020-10-26:
@kkumar23 for initial setup of docker with wsl,  it needs to connect to the docker engine running under windows.
Perhaps this will help:
https://blog.jayway.com/2017/04/19/running-docker-on-bash-on-windows/
Check section 2, I guess that might solve your problem.

kkumar23, 2020-10-26:
@coolbreeze413 : Thanks I am able to run now.

closed 2020-10-26 by kkumar23