QuickLogic-Corp/quicklogic-fpga-toolchain #39: Large number of warnings from ql_symbiflow

murthyvedula, 2020-10-24:
ql_symbiflow outputs large number of warnings of the type:

Warning 2443: Model 'RAM_CE1_FE1_D0_PR0_WSA0_WSB0_VPR' input port 'FIFO_EN_1' has no timing specification (no clock specified to create a sequential input port, not combinationally connected to any outputs, not a clock input) 

This causes the CI builds to fail with the following error message:

The job exceeded the maximum log length, and has been terminated.

kkumar23, 2020-10-27:
@tpagarani : we had this issue . Any plans of fixing them ?

tpagarani, 2020-10-27:
@kkumar23 I believe there is a flag to turn off these warnings. Can you confirm with @kgugala ?

kkumar23, 2020-10-28:
@kgugala Do we have any flag for these warnings ?

kkumar23, 2020-11-20:
@kgugala : Any flag you can suggest ?

kgugala, 2020-11-20:
you need to add `warn_model_missing_timing` to a list provided with the `suppress_warnings` flag in VPR

kkumar23, 2020-11-24:
@murthyvedula : Please verify this fix in the symbiflow v1.3.0

closed 2020-11-26 by kkumar23