QuickLogic-Corp/quicklogic-fpga-toolchain #40: Create account on hub.docker.com for Docker container images

Thirsty2, 2020-10-29:
Publishing the images on Docker Hub allows pulling them without using longer (sometimes awkward) URLs.  If we publish an image named symbiflow-ql to Docker hub, anyone (or any machine building downstream docker images) can use simple URLs like:

docker run -it --rm -v $(pwd):/root/work symbiflow-ql:1.3.0 bash

or at the start of a Dockerfile:

FROM symbiflow-ql:1.3.0

It is possible to use URLs like this already, but only if a user has locally tagged an image as symbiflow-ql:1.3.0.  In order to use a prebuilt package from github, you need a longer and awkward url that is more difficult to remember, and you need to be authenticated as a github user.  The above become:

docker run -it -rm -v $(pwd):/root/work docker.pkg.github.com/quicklogic-corp/quicklogic-fpga-toolchain/symbiflow-ql:1.3.0 bash

and 

FROM docker.pkg.github.com/quicklogic-corp/quicklogic-fpga-toolchain/symbiflow-ql:1.3.0

and these only work for users who are authenticated with github.

kkumar23, 2020-12-01:
@Thirsty2 : right will create a separate account. Will plan for that. Is docker supported in RHEL ? I do not see docker install on RHEL.https://docs.docker.com/engine/install/ . On Ubuntu I could run.

Thirsty2, 2020-12-01:
I believe Docker is supported on recent versions of RHEL.  I have not tried it personally.

Doug

whatnick, 2021-01-25:
This can be easily done in GitHub actions with this pre-built one : https://github.com/marketplace/actions/build-docker-images-using-cache

Better still we can push to all registries.

kkumar23, 2021-02-25:
We are not planning to publish docker images for now.

closed 2021-02-25 by kkumar23