QuickLogic-Corp/quicklogic-fpga-toolchain #41: The symbiflow-ql:1.3.0 package was built with the 1.2.0 installer

Thirsty2, 2020-10-29:
We need to create a new tagged package, perhaps something like 1.3.0.0 to trigger a new build of the latest tagged symbiflow-ql package.  The current release was tagged before changing the Dockerfile to use the 1.3.0 installer.  It also contains permissions that only allow the root user to run the tools.  If we create a new tag, the package will be re-built with the current Dockerfile that uses the 1.3.0 installer, and changes the permissions to 755 so anyone can run the symbiflow-ql tools.
 
https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/packages/449834

We should change the github action build to automatically inject the --build-arg TAG to match the current tag during the package build so we don't need to change the default in the Dockerfile.  Perhaps a non-working default tag like v0.0.0 as suggested by coolbreeze413 is a good idea, as it would only build if someone injects a proper tag or manually tweaks the default on their machine.  Perhaps this deserves its own separate issue entry.

Thirsty2, 2020-11-03:
The following pull request should prevent this issue going forward.
https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/pull/42

After the pull request is merged, one way to fix the package is to delete the current tag with 
git push --delete origin tagName

Then re-create tag v1.3.0 and push it, which will trigger a rebuild of the package and update it with the version built from the 1.3.0 installer.

kkumar23, 2020-11-19:
@Thirsty2 : have re-created the release tag 1.3.0

Thirsty2, 2020-11-20:
I tested the docker container image packages, and they both work.  Thanks for re-publishing the 1.3.0 tag.

closed 2020-11-20 by Thirsty2