QuickLogic-Corp/quicklogic-fpga-toolchain #43: Conda installer is not reproducible

whitequark, 2020-11-07:
The last step in the conda installer is to install a pip package from a git repository without specifying a particular revision. This means that, even if you might expect that you will be able to run `sh Symbiflow_v1.3.0.gz.run` in the future and get the exact same packages installed as when you ran it in the past, this is not the case.
```
pip install git+https://github.com/QuickLogic-Corp/quicklogic-fasm
```

kkumar23, 2020-11-19:
@kgugala if this is right, should we specify any particular version ?

kgugala, 2020-11-19:
in 1.3 we can freeze it on current master's HEAD. Simply change the line to:

```
pip install git+https://github.com/QuickLogic-Corp/quicklogic-fasm@318abca
```

kkumar23, 2020-11-20:
@whitequark : updated with the suggestions from Karol. Now it should refer to a fixed version

whitequark, 2020-11-20:
I've checked the script and I believe this issue is fixed.

closed 2020-11-20 by whitequark