QuickLogic-Corp/quicklogic-fpga-toolchain #44: Conda installer must not mangle ~/.condarc or use global channels

whitequark, 2020-11-07:
Mangling user configuration without asking (*especially* setting `always_yes`) is user-hostile. Using system/user channels in the Symbiflow installer means the packages that get actually installed are potentially broken and definitely nonreproducible. The following patch to the installer script fixes both issues:
```patch
--- conda_build_install_package.sh.orig	2020-11-07 16:48:01.376163599 +0000
+++ conda_build_install_package.sh	2020-11-07 16:47:58.688148160 +0000
@@ -19,17 +19,15 @@
 wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -O conda_installer.sh
 bash conda_installer.sh -b -p $INSTALL_DIR/conda && rm conda_installer.sh
 source "$INSTALL_DIR/conda/etc/profile.d/conda.sh"
-conda config --set always_yes yes --set changeps1 no
-conda config --add channels conda-forge
-conda config --add channels quicklogic-corp/label/ql
-conda update -y -q conda
+CONDA_FLAGS="-y --override-channels -c defaults -c conda-forge"
+conda update $CONDA_FLAGS -q conda
 wget -qO- https://quicklogic-my.sharepoint.com/:u:/p/kkumar/EQro5PGhiMpBhlieiFNoiuwBdQ-lpi-3VjqvRmYow54hEA?download=1 | tar -xJ -C $INSTALL_DIR
-conda install -c quicklogic-corp/label/ql yosys="0.8.0_0002_gc3b38fdc 20200901_073908" python=3.7
-conda install -c quicklogic-corp/label/ql yosys-plugins="1.2.0_0009_g9ab211c 20201001_121833"
-conda install -c quicklogic-corp/label/ql vtr="v8.0.0_rc2_2894_gdadca7ecf 20201008_140004"
-conda install -c quicklogic-corp iverilog
-conda install -c tfors gtkwave
-conda install -y make lxml simplejson intervaltree git pip
+conda install $CONDA_FLAGS -c quicklogic-corp/label/ql yosys="0.8.0_0002_gc3b38fdc 20200901_073908" python=3.7
+conda install $CONDA_FLAGS -c quicklogic-corp/label/ql yosys-plugins="1.2.0_0009_g9ab211c 20201001_121833"
+conda install $CONDA_FLAGS -c quicklogic-corp/label/ql vtr="v8.0.0_rc2_2894_gdadca7ecf 20201008_140004"
+conda install $CONDA_FLAGS -c quicklogic-corp iverilog
+conda install $CONDA_FLAGS -c tfors gtkwave
+conda install $CONDA_FLAGS make lxml simplejson intervaltree git pip
 conda activate
 pip install python-constraint
 pip install serial
```

mithro, 2020-11-08:
I don't know where the `conda_build_install_package.sh` script comes from. That script seems to have lost the steps which made conda use a local only `.condarc` file.

Those changes where super important and they are what made doing the `conda config` commands okay, as it was not touching the users `~/.condarc` or anything in the home directory. The CI should probably be checking that the `~/.conda*` directories are not touched during an install.

whitequark, 2020-11-08:
> I don't know where the `conda_build_install_package.sh` script comes from.

I unpacked the `Symbiflow_v1.3.0.gz.run` installer downloaded from the releases page of this repo using a command line option.

kkumar23, 2020-11-08:
@whitequark : Thanks for the fix. I shall make these changes. @mithro : conda_build_install_package.sh is not part of the repo. We wanted to keep it simple for a user to just run simple script and install symbiflow. Open to suggestions on this.

mithro, 2020-11-08:
@kkumar23 -- It should be in some repo somewhere...

kkumar23, 2020-11-09:
@mithro Yes its added in symbilfow-arch-defs repo (quicklogic-upstream-rebase branch).

kkumar23, 2020-11-20:
@whitequark : suggested changes are incorporated.

whitequark, 2020-11-20:
I've checked the latest installer and I believe this issue is fixed.

closed 2020-11-20 by whitequark