QuickLogic-Corp/quicklogic-fpga-toolchain #45: Conda environment should provide a single activation entry point

whitequark, 2020-11-07:
To build gateware with nMigen using the Quicklogic toolchain, one might naively think that the steps in README would work. However, that would actually override the Python interpreter used in that shell, and nMigen itself would no longer be in it. Because of this, nMigen build scripts source the path in the `NMIGEN_ENV_QLSymbiflow` environment variable, so that only the build script itself runs in the conda environment.

Currently, the README suggests the following:
```shell
export INSTALL_DIR="specify the installpath"
export PATH="$INSTALL_DIR/install/bin:$INSTALL_DIR/install/bin/python:$PATH"
source "$INSTALL_DIR/conda/etc/profile.d/conda.sh"
conda activate
```

The developer would have to put a file with these commands somewhere and then export the path to it `NMIGEN_ENV_QLSymbiflow`. That's not very convenient. It would be much better if Symbiflow provided such a file out of the box, just like ISE, Vivado, Diamond, etc do.

I.e. I would like the following command to be sufficient for setting up the environment:
```
source /install/path/conda/something/whatever.sh
```

There no longer seems to be any way to source and activate a conda environment using conda commands alone, so this is something Symbiflow should provide on its own.

kkumar23, 2020-11-20:
@whitequark : user needs to provide atleast the information of INSTALL_DIR, the rest 3 lines can be added in a script and can be sourced. Its good to have feature.

whitequark, 2020-11-20:
You should put `INSTALL_DIR` and the other necessary lines into a script when installing the toolchain so there is only a single file to be sourced and it doesn't need any environment variables.

kkumar23, 2021-02-25:
Installation has now setup.sh. Once installed user needs to source setup.sh file to activate the environment.

closed 2021-02-25 by kkumar23