QuickLogic-Corp/quicklogic-fpga-toolchain #46: OpenOCD script for IOMUX configuration is concatenated incorrectly

mkurc-ant, 2020-11-17:
In toolchain version 1.3.0 when `ql_symbiflow` is run with the `-dump openocd` option it generates the `top.openocd` file that contains a script that programs the FPGA and sets up the IOMUX.

However the IOMUX part gets concatenated incorrectly i.e. ends up outside the `program_bitstream` function. Example:
```
    ...
    mww 0x40004020 0x0000020a
    echo "Bitstream loaded successfully!"
}
    mww 0x40004C58 0x00000030
    mww 0x40004C78 0x00000030
    mww 0x40004D80 0x40400000
    mww 0x40004D84 0x00000000
```

The last 4 `mww` should be placed before the closing curly bracket.

The issue is in `symbiflow_write_openocd` wrapper script that does a blind concatenation:
```
echo "Generating openocd file"
cat ${TOP_F}.bit.openocd ${TOP_F}_iomux.openocd > ../${TOP_F}.openocd
```

kkumar23, 2020-11-23:
@coolbreeze413 : I hope recent PR to fpga toolchain has this fix?

coolbreeze413, 2020-11-23:
Yes, this should get fixed with the changes in the PR.

coolbreeze413, 2020-12-14:
With the v1.3.1 release, this should now be resolved.
Please re-open this issue, if the problem is seen again.
https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/releases/tag/v1.3.1

closed 2020-12-14 by coolbreeze413