QuickLogic-Corp/quicklogic-fpga-toolchain #47: No OpenOCD script for eFPGA RAM content initialization

mkurc-ant, 2020-11-17:
The toolchain stores the RAM initialization content in the generated `ram.mem` file. Unfortunately this content is not loaded to the chip by OpenOCD programming script (probably neither by JLINK script too).

To solve that the `bitstream_to_openocd.py` script should have a functionality to read `ram.mem` as well and incorporate its content into the programming sequence.

kkumar23, 2020-11-20:
@mkurc-ant We need to fix this in symbiflow-arch-defs right?

mkurc-ant, 2020-11-20:
@kkumar23 It depends. The Python script that converts bitstream to OpenOCD script is part of the `quicklogic-fasm` repo. It may be easier to extend its functionality so that it can read not only the bitstream but also `ram.mem`.

As far as I understand to load RAM content you have to set some control registers within the SoC. The process is connected with bitstream loading so I guess that augmenting `bitstream_to_openocd.py` is the way to go.

kkumar23, 2020-11-20:
Got it. You mentioned even the jlink script does not have the ram initialization , guess we need to add in bitstream_to_jlink.py as well.

kkumar23, 2020-11-23:
@coolbreeze413 : I believe recent PR to fpga toolchain has this fix?

coolbreeze413, 2020-11-23:
Yes, this should get fixed with the changes in the PR.
(JLink generation already had the ram initialization included previously, OpenOCD did not, which will get fixed in the PR)

coolbreeze413, 2020-12-14:
With the v1.3.1 release, this should now be resolved.
Please re-open this issue, if the problem is seen again.
https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/releases/tag/v1.3.1

closed 2020-12-14 by coolbreeze413