QuickLogic-Corp/quicklogic-fpga-toolchain #49: Dockerfile.use-installer broken

Thirsty2, 2020-11-22:
commit 7d0e3932e7492b4c309d366cd4c1e308688c0acc modified the ENV PATH statement:

+ENV PATH="${INSTALL_DIR}/quicklogic-arch-defs/bin:${INSTALL_DIR}/quicklogic-arch-defs/bin/python:$PATH"

A container built from the latest Dockerfile.use-installer with the 1.3.0 installer doesn't work properly.  Built with:
docker build -f Dockerfile.use-installer --build-arg TAG=v1.3.0 . -t symbiflow-ql:test

Then run with:
xhost +local:docker
docker run -it -e DISPLAY=$DISPLAY -v "/tmp/.X11-unix:/tmp/.X11-unix" symbiflow-ql:test bash

Then try:
source "$INSTALL_DIR/conda/etc/profile.d/conda.sh"
conda activate

#Execute the help command to display the help
ql_symbiflow -h

The ql_symbiflow -h command is not found.

kkumar23, 2020-11-23:
@Thirsty2  : The previous folder  "install" has been renamed to quicklogic-arch-defs, i have made these changes at docker install. Is there any other dependency ?

Thirsty2, 2020-11-23:
Does the change happen in an upcoming 1.4.0 installer?  It looks like the version 1.3.0 installer left the folder name as install.

Thirsty2, 2020-11-23:
Changing the PATH to the new name after using the 1.3.0 installer doesn't work.

Thirsty2, 2020-11-23:
It might be a good idea to build the installer automatically as a stage of the main Dockerfile which builds symbiflow-ql from source, but I don't know how Quicklogic builds their installers.  If we do this, we can ensure the paths are consistent.

kkumar23, 2020-11-23:
> Does the change happen in an upcoming 1.4.0 installer? It looks like the version 1.3.0 installer left the folder name as install.

@Thirsty2 : yes this will be the new name now instead of install. Well i did install again to check. It does not have the install name!

kkumar23, 2020-11-23:
> It might be a good idea to build the installer automatically as a stage of the main Dockerfile which builds symbiflow-ql from source, but I don't know how Quicklogic builds their installers. If we do this, we can ensure the paths are consistent.

Installer is just a script which has set of commands to be run.

Thirsty2, 2020-11-23:
Did you run the same installer that the Dockerfile.use-installer runs?  I tested that and it still uses install as the folder name.

Thirsty2, 2020-11-23:
The following line in Dockerfile.use-installer
RUN wget https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/releases/download/${TAG}/Symbiflow_${TAG}.gz.run
downloads the installer from here for the latest TAG of v1.3.0:
https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain/releases/download/v1.3.0/Symbiflow_v1.3.0.gz.run

kkumar23, 2020-11-23:
> The following line in Dockerfile.use-installer
> RUN wget [...]

Yes, and i could see the folder name(quicklogic-arch-defs) while installing. 
quicklogic-arch-defs/bin/
quicklogic-arch-defs/bin/symbiflow_generate_fasm2bels
quicklogic-arch-defs/bin/symbiflow_write_openocd

Thirsty2, 2020-11-23:
Should the steps I outlined in the first comment in this issue still work?  They didn't work for me with the new PATH pointing to symbiflow-arch-defs instead of install.

Thirsty2, 2020-11-24:
The v1.3.0 installer must have changed somehow, yet the version remains the same.  It worked 5 days ago when the symbiflow-ql:1.3.0 package was built.  You can verify this by running the pre-built github action packaged container and the steps reported as broken in the first comment in this issue above:

$ docker run -it -e DISPLAY=$DISPLAY -v "/tmp/.X11-unix:/tmp/.X11-unix" docker.pkg.github.com/quicklogic-corp/quicklogic-fpga-toolchain/symbiflow-ql:1.3.0 bash
root@9990912d04cd:/# source $INSTALL_DIR/conda/etc/profile.d/conda.sh
root@9990912d04cd:/# conda activate
root@9990912d04cd:/# ql_symbiflow -h

Below are the supported commands: 
 To synthesize and dump a eblif file:
	>ql_symbiflow -synth -src <source_dir path> -d <device> -P <package> -t <top> -v <verilog file/files> -p <pcf file>  
 To run synthesis, pack, place and route, generate bitstream:
	>ql_symbiflow -compile -src <source_dir path> -d <device> -P <package> -t <top> -v <verilog file/files> -p <pcf file>  
 To dump the jlink/post_verilog/header file: 
	>ql_symbiflow -compile -src <source_dir path> -d <device> -P <package> -t <top> -v <verilog file/files> -p <pcf file> -dump <jlink/post_verilog/header/openocd> 
Device supported:ql-eos-s3
Packages supported PD64,PU64,WR42 
-h

Note that in the symbiflow-ql:1.3.0 package created by the github action, there is still an install directory underneath $INSTALL_DIR:

root@9990912d04cd:/# cd $INSTALL_DIR/
root@9990912d04cd:/opt/symbiflow/eos-s3# ls -la
total 16
drwxr-xr-x  4 root root 4096 Nov 18 22:35 .
drwxr-xr-x  3 root root 4096 Nov 18 22:34 ..
drwxr-xr-x 21 root root 4096 Nov 18 22:36 conda
drwxrwxrwx  5 1000 1000 4096 Oct 25 05:26 install
root@9990912d04cd:/opt/symbiflow/eos-s3# 

Building locally with the same dockerfile today results in a broken container image.  The installer is the only thing that could have caused the breakage.

A new installer should have a different version tag, and be available at the URL for that new tag.

kkumar23, 2020-11-30:
@Thirsty2 : I ran the command docker build -f Dockerfile.use-installer --build-arg TAG=v1.3.0 . -t symbiflow-ql:test, and i see that the installer used is correct, but the folder quicklogic-arch-defs is not part of the docker image which is uploaded . 
In the INSTALL_DIR=/opt/symbiflow/eos-s3 . It has only the conda folder.  nor i see the install folder.

Thirsty2, 2020-11-30:
@kkumar23,
Dockerfile.use-installer never explicitly created quicklogic-arch-defs.  Instead, it depends on the installer to do that.  I ran the build command with an older  version of the dockerfile that worked fine when the 1.3.0 tagged release was created, and it behaves the same (broken) way.  That's how I know the installer is the problem.  The installer that was available for download (or something it in turn downloads) has changed since the 1.3.0 tagged release was created.  Everything worked at that time.  Is there a way to look at the installer script?  The Symbiflow_v1.3.0.gz.run file appears to be a self-extracting compressed binary.  If it downloads some dependencies you don't control, perhaps one of them changed?

kkumar23, 2020-11-30:
@Thirsty2 : The self extracting script run set of commands , here are those  :  
 wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -O conda_installer.sh
bash conda_installer.sh -b -p $INSTALL_DIR/conda && rm conda_installer.sh
source "$INSTALL_DIR/conda/etc/profile.d/conda.sh"
echo "include-system-site-packages=false" >> $INSTALL_DIR/conda/pyvenv.cfg
CONDA_FLAGS="-y --override-channels -c defaults -c conda-forge"
conda update $CONDA_FLAGS -q conda
curl https://storage.googleapis.com/symbiflow-arch-defs-install/quicklogic-arch-defs-63c3d8f9.tar.gz --output arch.tar.gz
tar -C $INSTALL_DIR -xvf arch.tar.gz && rm arch.tar.gz
conda install $CONDA_FLAGS -c quicklogic-corp/label/ql yosys="0.8.0_0002_gc3b38fdc 20200901_073908" python=3.7
conda install $CONDA_FLAGS -c quicklogic-corp/label/ql yosys-plugins="1.2.0_0009_g9ab211c 20201001_121833"
conda install $CONDA_FLAGS -c quicklogic-corp/label/ql vtr="v8.0.0_rc2_2894_gdadca7ecf 20201008_140004"
conda install $CONDA_FLAGS -c quicklogic-corp iverilog
conda install $CONDA_FLAGS -c tfors gtkwave
conda install $CONDA_FLAGS make lxml simplejson intervaltree git pip
conda activate
pip install python-constraint
pip install serial
pip install git+https://github.com/QuickLogic-Corp/quicklogic-fasm@318abca
conda deactivate

>>
The change in directory structure is at curl <quicklogic-arch-defs*>

Thirsty2, 2020-12-01:
It looks like the new installer requires curl and git, neither of which are installed in the ubuntu:20.04 base image or explicitly in the Dockerfile.use-installer.  We will need to add those to Dockerfile.use-installer.  Alternatively, and perhaps preferably, the installer would install all dependencies (including curl and git) that it needs, before it needs them.  

The Dockerfile.use-installer file already installs a few other missing dependencies with the following command:

RUN apt-get update && apt-get install -y \
    libcanberra-gtk-module \
    make \
    wget \
    xz-utils && \
    rm -rf /var/lib/apt/lists/*

We could add curl and git to the above list.

This new 1.3.0 installer (with curl and git) is definitely different than the 1.3.0 installer that was originally released.  That older version of the 1.3.0 installer worked fine on Nov 18, so it couldn't have required curl or git at that time.  How can we make sure that new versions of the installer don't get the same version number as an older version of the installer that worked?  This new installer should have been assigned a new version number like 1.3.1 or something similar.  

We also need to make sure the dependencies installed by the installer aren't changing as that could also break things.

kkumar23, 2020-12-01:
@Thirsty2 right. next release we are planning to take care of that.  Added the curl and git to the pre-requisite. Its fixed now.

closed 2020-12-01 by kkumar23