QuickLogic-Corp/quicklogic-fpga-toolchain #53: FASM error (antlr.py) "TypeError: encoding without a string argument"

tcal-x, 2021-01-22:
I just installed Symbiflow 1.3.1 from installer (Symbiflow_v1.3.1.gz.run), and attempted to run the counter\_16bit example as described in part 1) of https://github.com/QuickLogic-Corp/quicklogic-fpga-toolchain.   

I'm running on Ubuntu 20.04.

I encountered this:

```
Incr Slack updates 1 in 1.6095e-05 sec
Full Max Req/Worst Slack updates 1 in 3.732e-06 sec
Incr Max Req/Worst Slack updates 0 in 0 sec
Incr Criticality updates 0 in 0 sec
Full Criticality updates 1 in 3.2237e-05 sec
Writing Implementation FASM: top.fasm
The entire flow of VPR took 2.71161 seconds.
FASM extra: top_fasm_extra.fasm
cd build && symbiflow_write_bitstream -d ql-eos-s3_wlcsp -f top.fasm -P PD64 -b top.bit
Writing bitstream ...
Traceback (most recent call last):
  File "/home/tim/quicklogic/SymbiFlow/conda/bin/qlfasm", line 8, in <module>
    sys.exit(main())
  File "/home/tim/quicklogic/SymbiFlow/conda/lib/python3.7/site-packages/quicklogic_fasm/qlfasm.py", line 290, in main
    assembler.parse_fasm_filename(args.infile)
  File "/home/tim/quicklogic/SymbiFlow/conda/lib/python3.7/site-packages/fasm_utils/fasm_assembler.py", line 168, in parse_fasm_filename
    for line in fasm.parse_fasm_filename(filename):
  File "/home/tim/quicklogic/SymbiFlow/conda/lib/python3.7/site-packages/fasm/parser/antlr.py", line 104, in parse_fasm_filename
    parse_fasm.from_file(bytes(filename, 'ascii'), 0, callback, error_callback)
TypeError: encoding without a string argument
make: *** [Makefile.symbiflow:35: build/top.bit] Error 1
```
Let me know if there's any more information that you would find useful.

tcal-x, 2021-01-22:
More info: I also did a fresh install on a cloud VM with ubuntu 18.04, and I did not encounter the issue above.  I got the warning about fallback to using textx parser, but I did get a top.bit bitstream.

It is possible that the issue reported above, on my 20.04 laptop, is related to partial antlr installation.   I have tried to UNinstall the antlr packages, but fasm still wants to use antlr.   I'll try a couple more things.

tcal-x, 2021-01-23:
More findings:
On my system, in
```
    parse_fasm.from_file(bytes(filename, 'ascii'), 0, callback, error_callback)
```
`filename` is not a string, but a `<class 'pathlib.PosixPath'>`.   I can get it to work if I change `filename` to `str(filename)`:
```
    parse_fasm.from_file(bytes(str(filename), 'ascii'), 0, callback, error_callback)
```
But I don't understand why the behavior is different on my system;  I'm in the conda environment, so everything should be the same, right?

kishangondaliya, 2021-01-23:
@tcal-x I'm facing same issue with Ubuntu 20 fresh install.
Thank you for the work around. This is may be due to some recent update in packages which are prrobably incompatible/not updated for each other.

tcal-x, 2021-01-23:
P.S. `str(string)` will just give you `string` back, so the workaround I suggested above would not break things on other systems.

kkumar23, 2021-01-25:
I could not install and check, as i am getting the below error: 
CondaHTTPError: HTTP 500 INTERNAL SERVER ERROR for url <https://conda.anaconda.org/quicklogic-corp/label/ql/linux-64/repodata.json>
Elapsed: 00:10.439786
CF-RAY: 617162132cdf3c0d-BLR
A remote server error occurred when trying to retrieve this URL.

A 500-type error (e.g. 500, 501, 502, 503, etc.) indicates the server failed to
fulfill a valid request.  The problem may be spurious, and will resolve itself if you
try your request again.  If the problem persists, consider notifying the maintainer
of the remote server.

Will try out once the server is up.

coolbreeze413, 2021-01-26:
I tried to install antlr and then uninstall it inside the conda environment, and I get the same problem. However, the toolchain setup which was previously setup (before antlr installation) continues to work ok, and generates the bitstream using the textx parser (with the fallback warning).

change `filename` to `str(filename)` does work too. We will try to get the fasm and antlr dependencies sorted out in the installation procedure.