QuickLogic-Corp/quicklogic-fpga-toolchain #55: Building Docker from pre-built packages in conda fails

whatnick, 2021-01-25:
Running 

```
docker build . -t symbiflow-ql-slim-buster -f Dockerfile.use-installer
```

fails with

```
CondaHTTPError: HTTP 500 INTERNAL SERVER ERROR for url <https://conda.anaconda.org/quicklogic-corp/noarch/repodata.json>
Elapsed: 00:11.642342
CF-RAY: 616f1fea9d22fd46-SYD

A remote server error occurred when trying to retrieve this URL.

A 500-type error (e.g. 500, 501, 502, 503, etc.) indicates the server failed to
fulfill a valid request.  The problem may be spurious, and will resolve itself if you
try your request again.  If the problem persists, consider notifying the maintainer
of the remote server.

and

```
CondaHTTPError: HTTP 500 INTERNAL SERVER ERROR for url <https://conda.anaconda.org/tfors/linux-64/repodata.json>
Elapsed: 00:10.735852
CF-RAY: 616f203fcb90fd56-SYD

A remote server error occurred when trying to retrieve this URL.

A 500-type error (e.g. 500, 501, 502, 503, etc.) indicates the server failed to
fulfill a valid request.  The problem may be spurious, and will resolve itself if you
try your request again.  If the problem persists, consider notifying the maintainer
of the remote server.
```

The container builds successfully but does not have yosys tooling installed.

kkumar23, 2021-01-25:
@whatnick : Getting similar error with other repositories as well. i think its issue at the server side.

mounik92, 2021-01-25:
@whatnick : Facing same issue.

maresb, 2021-01-25:
For example, these channels are also affected:
https://github.com/plotly/plotly.py/issues/3046
https://github.com/pytorch/pytorch/issues/51029

maresb, 2021-01-25:
Seems to be fixed now.

whatnick, 2021-01-25:
I will rebuild this, however having pre-built ones in DockerHub would be great in lowering the barrier in getting this tooling and getting started. Docker being set as default container registry and all. I see the issue for this already. I happy to make a PR on the GitHub actions side with the push to DockerHub setup. I do it a lot for other projects.