QuickLogic-Corp/quicklogic-fpga-toolchain #56: conda.exe: cannot execute binary file

just4give, 2021-02-09:
Hi,
I am trying to install FPGA Toolchain on Mac Catalin 10.15.7. I created a folder ~/Documents/symbiflow and set thant as INSTALL_DIR

Then as per readme I executed 
`bash Symbiflow_v1.3.1.gz.run` from a folder when I downloaded the execution file. But I am getting following error. 

[image]

As a result conda sub-directories are not loaded in the installed directory. Any idea how to resolve this issue?

charkster, 2021-02-18:
I am seeing the same issue too (on Raspberry Pi OS).
```
--2021-02-18 16:23:23--  https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh
Resolving repo.continuum.io (repo.continuum.io)... 104.18.201.79, 104.18.200.79, 2606:4700::6812:c84f, ...
Connecting to repo.continuum.io (repo.continuum.io)|104.18.201.79|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh [following]
--2021-02-18 16:23:23--  https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh
Resolving repo.anaconda.com (repo.anaconda.com)... 104.16.130.3, 104.16.131.3, 2606:4700::6810:8203, ...
Connecting to repo.anaconda.com (repo.anaconda.com)|104.16.130.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 94235922 (90M) [application/x-sh]
Saving to: ?conda_installer.sh?

conda_installer.sh              100%[=======================================================>]  89.87M  4.87MB/s    in 21s     

2021-02-18 16:23:45 (4.36 MB/s) - ?conda_installer.sh? saved [94235922/94235922]

PREFIX=/home/pi/symbiflow_fpga/conda
Unpacking payload ...
conda_installer.sh: line 412: /home/pi/symbiflow_fpga/conda/conda.exe: cannot execute binary file: Exec format error
conda_installer.sh: line 414: /home/pi/symbiflow_fpga/conda/conda.exe: cannot execute binary file: Exec format error
./conda_build_install_package.sh: line 21: /home/pi/symbiflow_fpga/conda/etc/profile.d/conda.sh: No such file or directory
./conda_build_install_package.sh: line 24: conda: command not found
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 34.4M  100 34.4M    0     0  4530k      0  0:00:07  0:00:07 --:--:-- 4852k
```

kgugala, 2021-02-19:
hi @charkster 

you are installing `https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh` - as the name suggest this is conda binary for Linux on x86_64. Raspberry Pi has ARM CPU. It cannot work

In general there is no prebuilt toolchain for ARM. However, I've seen somebody running the toolchain on rPI, so it is doable - you'd need to rebuild the required tools - namely Yosys and VPR (the rest is Python or bash scripts).

Also the flow requires Python3 (Python2 is not supported)

charkster, 2021-02-19:
> you are installing `https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh` - as the name suggest this is conda binary for Linux on x86_64. Raspberry Pi has ARM CPU. It cannot work [...]
@kgugala Thanks for pointing this out. I will do the install on x86 Ubuntu.

coolbreeze413, 2021-02-20:
@charkster to use it on RPi, you would have to build the tools from source, as @kgugala mentioned.

We have done this, the instruction below should be able to help you if you want to go this route:
https://github.com/optimuslogic/pygmy-dev/blob/main/tools/tools-setup.txt

charkster, 2021-02-20:
@coolbreeze413  Thank you very much!!

charkster, 2021-02-21:
@coolbreeze413 I have everything working except qlfasm (which comes from the conda flow and installs the x86_64 version). Would you know how to build or download qlfasm? Thanks. I needed to update my pp3_cells_sim.v and pp3_cells_map.v files for the new definition of qlal4s3b_cell_macro in eos-s3.

gadepall, 2021-02-21:
Try this for RPi 3:
https://github.com/gadepall/vaman/blob/master/installation/refs/rpios_install.txt

charkster, 2021-02-21:
@gadepall Thanks!! that is just what I needed. Works great.