QuickLogic-Corp/quicklogic-fpga-toolchain #57: Install problem on Raspberry Pi OS

charkster, 2021-02-18:
Hi,
I'm trying to install on Raspberry Pi OS, and am getting this error during the installation. I have no problems until:
`writing manifest file 'pip-egg-info/fasm.egg-info/SOURCES.txt'`
```
Installing collected packages: pyyaml, future, iso8601, serial
Successfully installed future-0.18.2 iso8601-0.1.14 pyyaml-5.4.1 serial-0.0.97
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting git+https://github.com/QuickLogic-Corp/quicklogic-fasm@2312ff0
  Cloning https://github.com/QuickLogic-Corp/quicklogic-fasm (to revision 2312ff0) to /tmp/pip-req-build-ddZ4lu
  Did not find branch or tag '2312ff0', assuming revision or ref.
Collecting fasm-utils@ git+https://github.com/QuickLogic-Corp/quicklogic-fasm-utils#egg=fasm-utils from git+https://github.com/QuickLogic-Corp/quicklogic-fasm-utils#egg=fasm-utils (from quicklogic-fasm==0.0.1)
  Cloning https://github.com/QuickLogic-Corp/quicklogic-fasm-utils to /tmp/pip-install-8Fxzka/fasm-utils
Collecting fasm@ git+https://github.com/symbiflow/fasm#egg=fasm from git+https://github.com/symbiflow/fasm#egg=fasm (from quicklogic-fasm==0.0.1)
  Cloning https://github.com/symbiflow/fasm to /tmp/pip-install-8Fxzka/fasm
  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    /tmp/pip-build-env-c58qEW/lib/python2.7/site-packages/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /tmp/pip-install-8Fxzka/fasm/fasm/parser/antlr_to_tuple.pyx
      tree = Parsing.p_module(s, pxd, full_module_name)
    Compiling fasm/parser/antlr_to_tuple.pyx because it changed.
    [1/1] Cythonizing fasm/parser/antlr_to_tuple.pyx
    /tmp/pip-build-env-c58qEW/lib/python2.7/site-packages/setuptools/dist.py:485: UserWarning: The version specified ('UNKNOWN') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
      "details." % self.metadata.version
    running egg_info
    creating pip-egg-info/fasm.egg-info
    writing requirements to pip-egg-info/fasm.egg-info/requires.txt
    writing pip-egg-info/fasm.egg-info/PKG-INFO
    writing top-level names to pip-egg-info/fasm.egg-info/top_level.txt
    writing dependency_links to pip-egg-info/fasm.egg-info/dependency_links.txt
    writing entry points to pip-egg-info/fasm.egg-info/entry_points.txt
    writing manifest file 'pip-egg-info/fasm.egg-info/SOURCES.txt'
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-8Fxzka/fasm/setup.py", line 277, in <module>
        'install': InstallCommand,
      File "/tmp/pip-build-env-c58qEW/lib/python2.7/site-packages/setuptools/__init__.py", line 162, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
        dist.run_commands()
      File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
        cmd_obj.run()
      File "/tmp/pip-build-env-c58qEW/lib/python2.7/site-packages/setuptools/command/egg_info.py", line 296, in run
        self.find_sources()
      File "/tmp/pip-build-env-c58qEW/lib/python2.7/site-packages/setuptools/command/egg_info.py", line 303, in find_sources
        mm.run()
      File "/tmp/pip-build-env-c58qEW/lib/python2.7/site-packages/setuptools/command/egg_info.py", line 534, in run
        self.add_defaults()
      File "/tmp/pip-build-env-c58qEW/lib/python2.7/site-packages/setuptools/command/egg_info.py", line 570, in add_defaults
        sdist.add_defaults(self)
      File "/tmp/pip-build-env-c58qEW/lib/python2.7/site-packages/setuptools/command/py36compat.py", line 34, in add_defaults
        self._add_defaults_python()
      File "/tmp/pip-build-env-c58qEW/lib/python2.7/site-packages/setuptools/command/sdist.py", line 135, in _add_defaults_python
        build_py = self.get_finalized_command('build_py')
      File "/usr/lib/python2.7/distutils/cmd.py", line 312, in get_finalized_command
        cmd_obj.ensure_finalized()
      File "/usr/lib/python2.7/distutils/cmd.py", line 109, in ensure_finalized
        self.finalize_options()
      File "/tmp/pip-build-env-c58qEW/lib/python2.7/site-packages/setuptools/command/build_py.py", line 34, in finalize_options
        orig.build_py.finalize_options(self)
      File "/usr/lib/python2.7/distutils/command/build_py.py", line 46, in finalize_options
        ('force', 'force'))
      File "/usr/lib/python2.7/distutils/cmd.py", line 297, in set_undefined_options
        src_cmd_obj = self.distribution.get_command_obj(src_cmd)
      File "/usr/lib/python2.7/distutils/dist.py", line 846, in get_command_obj
        cmd_obj = self.command_obj[command] = klass(self)
      File "/tmp/pip-build-env-c58qEW/lib/python2.7/site-packages/setuptools/__init__.py", line 180, in __init__
        _Command.__init__(self, dist)
      File "/usr/lib/python2.7/distutils/cmd.py", line 64, in __init__
        self.initialize_options()
      File "/tmp/pip-install-8Fxzka/fasm/setup.py", line 206, in initialize_options
        super().initialize_options()
    TypeError: super() takes at least 1 argument (0 given)
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-8Fxzka/fasm/
./conda_build_install_package.sh: line 38: conda: command not found
```
Could this be related to the default python being 2.7 ?
Thanks

charkster, 2021-02-18:
I am also seeing issue #56 . Not sure if this is related.

charkster, 2021-02-19:
It looks like only x86 binaries are supported for this install flow so I will close the issue.

closed 2021-02-19 by charkster

coolbreeze413, 2021-02-20:
@charkster this might help you to get the toolchain running with the RPi:

https://github.com/optimuslogic/pygmy-dev/blob/main/tools/tools-setup.txt