QuickLogic-Corp/quicklogic-fpga-toolchain #58: Install problem on Ubuntu 20.04

charkster, 2021-02-18:
Tried doing the install on Ubuntu 20.04 ( got a little further compared to Raspberry Pi OS #57 ).
It looks like a file version issue (complains about `UNKNOWN` version).

```
Successfully installed future-0.18.2 iso8601-0.1.14 pyyaml-5.4.1 serial-0.0.97
Collecting git+https://github.com/QuickLogic-Corp/quicklogic-fasm@2312ff0
  Cloning https://github.com/QuickLogic-Corp/quicklogic-fasm (to revision 2312ff0) to /tmp/pip-req-build-fde_skq3
  Running command git clone -q https://github.com/QuickLogic-Corp/quicklogic-fasm /tmp/pip-req-build-fde_skq3
  WARNING: Did not find branch or tag '2312ff0', assuming revision or ref.
  Running command git checkout -q 2312ff0
  Running command git submodule update --init --recursive -q
Collecting fasm@ git+https://github.com/symbiflow/fasm#egg=fasm
  Cloning https://github.com/symbiflow/fasm to /tmp/pip-install-qp40rf0a/fasm_cfb329fd84b64cfabe7b8d609d3a7ac0
  Running command git clone -q https://github.com/symbiflow/fasm /tmp/pip-install-qp40rf0a/fasm_cfb329fd84b64cfabe7b8d609d3a7ac0
  Running command git submodule update --init --recursive -q
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting fasm-utils@ git+https://github.com/QuickLogic-Corp/quicklogic-fasm-utils#egg=fasm-utils
  Cloning https://github.com/QuickLogic-Corp/quicklogic-fasm-utils to /tmp/pip-install-qp40rf0a/fasm-utils_3d65fe6d432c400a8bece711c4f44b29
  Running command git clone -q https://github.com/QuickLogic-Corp/quicklogic-fasm-utils /tmp/pip-install-qp40rf0a/fasm-utils_3d65fe6d432c400a8bece711c4f44b29
Collecting textx
  Downloading textX-2.3.0-py2.py3-none-any.whl (72 kB)
     |████████████████████████████████| 72 kB 957 kB/s 
Collecting Arpeggio>=1.9.0
  Downloading Arpeggio-1.10.1-py2.py3-none-any.whl (54 kB)
     |████████████████████████████████| 54 kB 3.1 MB/s 
Building wheels for collected packages: quicklogic-fasm, fasm, fasm-utils
  Building wheel for quicklogic-fasm (setup.py) ... done
  Created wheel for quicklogic-fasm: filename=quicklogic_fasm-0.0.1-py3-none-any.whl size=2799929 sha256=e8401ac4cd24ef1e8e989c0103bd6f1f4546ca2939d6e8dd21ec0d8f607f16dd
  Stored in directory: /tmp/pip-ephem-wheel-cache-xw5xe2m0/wheels/6f/3b/9b/a098bf81f2c1ac3f2ea8261fbe47c225f302863a2007223a6f
  Building wheel for fasm (PEP 517) ... done
  Created wheel for fasm: filename=fasm-UNKNOWN-cp37-cp37m-linux_x86_64.whl size=16404 sha256=b830d797c208c6dadf403c577179ab2b6bc0c9f777ec3dbbe7054904537bd742
  Stored in directory: /tmp/pip-ephem-wheel-cache-xw5xe2m0/wheels/ab/55/95/7041cdff027c4fcab7c0c11283e17d227dc493af697702d238
  WARNING: Built wheel for fasm is invalid: Metadata 1.2 mandates PEP 440 version, but 'UNKNOWN' is not
  Building wheel for fasm-utils (setup.py) ... done
  Created wheel for fasm-utils: filename=fasm_utils-0.0.1-py3-none-any.whl size=9270 sha256=560d9bfd0ca709f2f99b1afd92951655c61abc7b7c92146ba50198aea5da60cf
  Stored in directory: /tmp/pip-ephem-wheel-cache-xw5xe2m0/wheels/7d/d3/59/f76589923da0abba5a47e629ee9881eadec8c002eacfe7d53b
Successfully built quicklogic-fasm fasm-utils
Failed to build fasm
ERROR: Could not build wheels for fasm which use PEP 517 and cannot be installed directly
```

DavidGoedicke, 2021-02-19:
Same  error here too.

Running on x86_64.

charkster, 2021-02-19:
Looks like a problem on the SymbiFlow/fasm side.
https://github.com/SymbiFlow/fasm/issues/60
I'll try again when they close their issue.

charkster, 2021-02-20:
Looks like SymbiFlow/fasm#60 was the problem. After they fixed it I was able to install and run the examples without any issues.

closed 2021-02-20 by charkster