QuickLogic-Corp/quicklogic-fpga-toolchain #72: Still supported?

xyphro, 2024-02-22:
I also got issues installing the toolchain. Only later I noticed that the last push was done 3 years ago. Is this toolchain still supported?
Is the product still supported? It looks great, but without toolchain it is useless :-(

QL-Robert, 2024-03-11:
I assume that you are looking for support for EOS S3 or is it another FPGA device?  
EOS S3 FPGA toolchain support has been incorporated into the f4pga branch details can be found at https://f4pga-examples.readthedocs.io/en/latest/getting.html

If you are looking for some example project then I would suggest the repo at https://github.com/search?q=qorc-onion-apps&type=repositories

Entropy512, 2024-06-05:
I'm looking through the F4PGA documentation, and it states here - https://f4pga.readthedocs.io/en/latest/f4pga/Usage.html :
"Quicklogic EOS-S3 (currently unsupported, provided only for development purposes)"

Is this documentation out of date, or are there unmerged pull requests that users need to be aware of?

QL-Robert, 2024-06-17:
As mentioned the QuickLogic device support has been merged into the f4pga release, is there issues you are facing with this release?

Entropy512, 2024-07-07:
My issues have already been stated - F4PGA's documentation lists Quicklogic support as "Quicklogic EOS-S3 (currently unsupported, provided only for development purposes)" per the documentation link I have already provided.

Please explain this discrepancy?